Mabo-IoT/mission-brief #41: AVL_alarms采集程序watchman有些问题

HeathKang, 2018-05-16:
## issues：
- 无法正常运行，显示json无法加载
- 检查发现```recorder```里的记录json为空，造成无法读取的错误

## solutions
- 临时删除record json，可以使程序正常运行
- 观察、复现bug，查找真正原因
- 将```conf``` console改为false，获取程序的file日志输出，更好的发现bug

HeathKang, 2018-05-16:
## update：
重点查看一下file_watcher 的set_seek()和get_seek()函数，看是否是win和linux系统对"/" "\"处理不一致导致的bug

Pyligo, 2018-05-16:
原因分析相关：因程序在4月11号左右就部署上去了，当时程序成功无误，所以我判定问题不在于win和Linux系统对“/”处理不一致，有可能是因为所读取的messages.jrn文件有定期清理机制，待检验；
现解决方案：增加了程序针对read_recorder中文件内容为空时重新读取日志的操作。
这样是否可行？

Pyligo, 2018-05-17:
原因分析相关：经检验，messages.jrn文件清理并不会导致recorder里存放的指针为空，但出现的log信息一样

Pyligo, 2018-05-23:
5月23日最新检测情况
经检验，挂载不稳定会同样会使之前存放的指针失效
检测方式：将要读取的文件放置在U盘，读取后插拔U盘，watchman报同样的错误
解决方案：暂不处理指针的问题，查看watchman对于路径失效的处理机制，改善相应模块

Pyligo, 2018-05-25:
5月25最新检测情况：
经检验， 程序跑时记录的指针无效
检测方式：程序正常跑时cut掉程序，重新运行程序，报错无法正常读取该文件
解决方案：查看watchman指针记录方式，改善相应模块

Pyligo, 2018-05-29:
问题发现：watchman程序异常中断后，不再正常读取之前未读取的内容
进一步检测：watchman异常中断后，如果往文件中写入新的内容时，程序会不会继续正常运行

Pyligo, 2018-05-29:
已修复以上bug，主要改动在get_seek,set_seek以及_watcher函数中，修改为每读一行就记录一次指针，以防程序异常中断导致数据丢失。详细代码请在reset_watchman's_file_watcher中查看

Pyligo, 2018-05-29:
log2上已统一修改，需要观察追踪修改结果

Pyligo, 2018-05-31:
log1 log2 均已修改，包括非挂载方式的1M03设备，使用的均是已修改好的watchman

Pyligo, 2018-06-08:
新bug：messages.prn文件有备份机制，导致记录的指针大于文件大小
修正：当检测到seek大于文件大小时，就讲指针置0
除1M03尚未修正，其它已修正完成，需要进一步观察追踪修改情况

closed 2018-06-20 by Pyligo

Pyligo, 2018-07-18:
1M03 修正完成。