Mabo-IoT/mission-brief #86: filebeat使用部署

Pyligo, 2018-06-28:
- [ ] 首先在windows上使用，并写好一个插件，功能要求能替代watchman

- [ ] 做好必要的测试，确定基本调试部署步骤方法，环境要求

- [ ] 在centos7环境安装使用，调整调试部署步骤方式，环境配置，写成可使用文档
- [ ] 写各种设备的插件

Pyligo, 2018-07-05:
实现了第一步，但速度太慢，下一步换Linux系统考虑使用协程

majj, 2018-07-14:
官方不建议采用共享目录形式部署beat，建议制作一键安装，在操作系统重做后可快速部署。

[filebeat faq](https://www.elastic.co/guide/en/beats/filebeat/current/faq.html)

# Can’t read log files from network volumes?edit
We do not recommend reading log files from network volumes. Whenever possible, install Filebeat on the host machine and send the log files directly from there. Reading files from network volumes (especially on Windows) can have unexpected side effects. For example, changed file identifiers may result in Filebeat reading a log file from scratch again

Pyligo, 2018-07-18:
插件正确性检测：

- [ ] 将时间戳的修改添加进去；
- [ ] 针对mts日志log信息获取不完整问题，需保证filebeat正则语句的正确性，是否会产生冲突，比如以小括号开头，log信息中是否也会出现小括号。插件里需要设置相关操作来检验filebeat的正则匹配。
- [ ] status通过状态机计算

majj, 2018-07-20:
^\( 是句首括号，msg总的括号不在句首。
上一条检测的结论是什么？

Pyligo, 2018-07-20:
时间戳的修改方式——一个文件对应一个seq，方法是可行的。

Pyligo, 2018-07-20:
第二个是插件的问题，mts插件内部有检测接收的msg信息是否符合要处理的格式，未考虑msg双引号内会出现回车符的情况

majj, 2018-07-20:
多行日志，肯定是双引号内有回车的。单行日志行之间是\r\n, 多行日志用\n分割

Pyligo, 2018-07-24:
插件各种处理已修改测试完毕，下一步将程序部署在一台中间机对多台设备日志进行分析处理。

Pyligo, 2018-08-25:
status sender需要使用框架来读取不同设备的状态并上传，升级status sender

Pyligo, 2018-08-27:
status sender 已改好，改写的watchman，不同的设备status开不同的线程上传。

HeathKang, 2019-05-27:
已完成

closed 2019-05-27 by HeathKang